NiryoRobotics/niryo_one_ros #2: Include Gripper in Robot Model

smaassen, 2018-01-27:
Can you provide, at least for the standard gripper, an urdf model + moveIT configuration (including IKFast)? In my eyes, this is quite essential if you want to use the MoveIT library with your robot in ROS. Any task where you want to pick up or place an object for example.

Thanks

renarded, 2018-01-29:
Hi,

Moveit! motion planning is only used for the 6 axis of the robot. Not for any tool. The reasons are : 
- there are many tools that have different shapes and behaviors (not only grippers).
- adding grasping functionality inside Moveit! would require to add each object to pick in the 3D planning scene. This would require much more computational power (Niryo One is powered by Raspberry Pi 3) and a lot of configuration from the user. We want things to be simple and easy.
- It's much easier to add/switch a tool that way. And anyone can create a new gripper or other tool with instructions [here](https://github.com/NiryoRobotics/niryo_one_ros/tree/master/niryo_one_tools)
- Also, the state of the art in grasping is not much developed compared to 6 axis motion planning

So the planning point used for Niryo One is always the end of the robot "hand". From that point, you can use any tool you want with a separate way of controlling it.

I hope this answers your questions. The reasons might not seem very obvious but in the end this is the best solution (for now).

smaassen, 2018-01-29:
Hey, 

I see your points and they make a lot of sense to me, however i have a few further question:

- concerning the grasping functionality, you could do the planing (i.e. the collision avoidance of scene objects) on a seperate machine right?
- Despit your points made, it would be nice to have the transform to the middle of the grippers in order to control the gripper to an desired object. I saw that you foresaw this in your urdf's by defining a distance variable for the last gripper_link, which at the moment is coincident with hand_link. I tried to change this, since it would make life much easier for grasping something. I recompiled the ikfast library and updated the moveit config, after changing the urdf but i still can not get the inverse kinematic solver to work. Am i missing something here? Could you maybe provide a short list of things I'd have to consider when offsetting the last frame.

Thanks a lot for the fast repsonse.

renarded, 2018-01-30:
- Yes you could use another computer (non ARM) to do the planning with Moveit!. You'll be able to get full advantages of all the 3D tools (planning scene with obstacles, etc). This will require some additional work.
- Well I haven't tested yet the IKfast plugin with an additional link for the gripper. Did you just recompile it ? You may need to redo the entire package (tutorial [here](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/ikfast_tutorial.html)). 

We added a gripper_link at the beginning that we are not using yet, and from the current evolution of things, I don't think we'll change the way of controlling tools.

Feel free to share your progress ! (we'll open a forum on our website soon so you can easily discuss with other Niryo One users)

smaassen, 2018-01-31:
Since I am doing a lot of computation on another machine anyways, i need to get an interface to my own rosnode running on that machine. Ideally I would like to use moveIT in that rosnode and make the planning on that computer and only send the trajectories to be executed to the robot. How would you attack that problem (i.e what is the additional work you related to)? 
Is it possible to somehow remap the moveIT nodes and topics? Or would you suggest running over your robot commander?

Sorry for the questions but it is a bit hard to understand your full code structure from the documentation here

renarded, 2018-02-05:
Well basically, you just need to keep [this controller configuration](https://github.com/NiryoRobotics/niryo_one_ros/blob/master/niryo_one_moveit_config/config/controllers.yaml) file in your moveit_config package. The hardware controller (in driver) has a actionlib server, and this config file  allows Moveit! to automatically send the computed trajectory to the controller (trhough the actionlib server). 

If you are working with a remote computer, you just need to make sure that messages and services can be sent and received from all your ROS devices, and then the interface between Moveit! and the controller doesn't need any change.

You can also remap topics, this is a ROS functionality. For that please check out ROS wiki.

---

The basic architecture for a command is "robot_commander" -> "Moveit!" -> "driver" (with controller)

The robot commander was developed to provide a simple interface for all user commands, with params validation, concurrency check, etc. You can skip this and directly use Moveit! with the movegroup class. (python or cpp)

And no worries, I know that the full code can be quite hard to understand, even with the documentation :) This is also due to some other ROS packages that are used but already installed, so you can't see their code here, thus there are a lot of bridges missing if you only read this repo.

closed 2018-09-11 by renarded