NiryoRobotics/niryo_one_ros #5: Problem running niryo_one_moveit_config demo.launch

eyalu, 2018-02-14:
rviz crashes when I run:
`roslaunch niryo_one_moveit_config demo.launch`

This is from an up to date source, with no modifications (and with the correct setup.bash). I can't find any reason for the crash in the error files, can you please assist?

The files are attached:
[crash.tar.gz](https://github.com/NiryoRobotics/niryo_one_ros/files/1724002/crash.tar.gz)

renarded, 2018-02-14:
Hi,

Thanks for having attached the log files. 

One possibility I think of is some recent ROS kinetic updates that are breaking compatibility. We'll look into it and get back to you soon.

eyalu, 2018-02-14:
Hello,

Look no more - indeed updating rviz solved the issue. Sorry to bother you, should have investigated further before opening this issue.
 
To be clear - **this is not a Niryo related problem**.

Just for reference for future developers who experience the same problem:
The crash as shown when debugging:

> Thread 1 "rviz" received signal SIGSEGV, Segmentation fault.
> 0x00007ffff74a85f5 in QWidgetPrivate::init(QWidget*, QFlags<Qt::WindowType>) ()
>    from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
> 

This is a problem that has to is related to Qt4/Qt5 and OpenCV dependencies that were resolved by the rviz developers in May 2017 (reference: https://github.com/ros-visualization/rviz/issues/1026). This will happen in Ubuntu 16.04 too.

closed 2018-02-14 by eyalu