NiryoRobotics/niryo_one_ros #20: Error running simulation mode.

SantiagoCanete, 2018-11-30:
Hello,

I am trying to use the API in simulation mode. I am running the following command to start the simulation:

`roslaunch niryo_one_bringup desktop_rviz_simulation.launch`

For some reason, it is giving me a set of errors and the 3D model is not rendering correctly. I am attaching the log files.

[e8b3d0d2-f4e5-11e8-a2d8-4ccc6af183fb.zip](https://github.com/NiryoRobotics/niryo_one_ros/files/2635010/e8b3d0d2-f4e5-11e8-a2d8-4ccc6af183fb.zip)

Thank you in advance.

renarded, 2018-12-03:
Hi,
I have checked your log and the problem is due to an issue with TF not working.

Could you precise:
- what is your environment
- which version of ROS are you using
- the exact steps you did to get to this point

Thanks.

SantiagoCanete, 2018-12-03:
Hello,

Thank you for the quick response! I am using an Ubuntu shell in Windows 10. The version is Ubuntu 16.04. My ROS version is Kinetic, I did a full desktop install and also installed all the packages mentioned in the niryo_one_ros installation guide. I am using Xming as the display server between Ubuntu and Windows. What does TF stand for?

Thank you in advance.

renarded, 2018-12-03:
See [here](http://wiki.ros.org/tf) for a complete tf explanation.

Tf will be used internally for Moveit!, also for publishing on the /joint_state topic, and with Niryo One we also use it to get the position and euler rotation of the tool center point.

So, if Tf is not working properly, the robot will not work because it has no way to keep its state. And the problem here is not related to Niryo One's code but rather the configuration of Tf in your environment. Something is failing inside Tf.

One point that maybe can help you: Once I had an issue (on a fresh install) with Tf not working. The log told me about an internal error about the Numpy Python package.

You could try to run "sudo -H pip install numpy" and see if it works (this command can be quite long and take a few minutes). Nothing guaranteed but you might get lucky :)

Also I've never tried running a Ubuntu shell in Windows 10. Maybe you can do some researches based on that and the Tf ROS functionality (also with robot_state_publisher)

And if you can't manage to make it work, please also test on a Ubuntu virtual machine or dual boot directly to see if you're able to get the simulation working.

SantiagoCanete, 2018-12-03:
It did work on an Ubuntu machine. It is most likely a problem with the X server. For what I have been reading it is hard to get rviz working properly through Xming.

Thank you for your help.

SantiagoCanete, 2018-12-05:
Hello,

We are very interested in getting this to work in an Ubuntu shell in Windows. 
When running in display mode /tf and /tf_static are being published correctly. But, when running in simulation mode /tf_static still works but not /tf. 

In display mode:
[image]

In Simulation mode:
[image]

[image]

[image]

Therefor I do not believe it is a problem with the tf package. Is there anyway you could help me? We will be using the niryo_one system for a class next year and we need students to be able to simulate their code without needing a full Ubuntu machine. 

Thank you again.

renarded, 2018-12-07:
So, if I understand well, the `roslaunch niryo_one_description display.launch` command is working correctly with /tf, and not `roslaunch niryo_one_bringup desktop_rviz_simulation.launch`.

The problem could come from either the /robot_state_publisher node (which is in charge of publishing to /tf), or the failure of the joint_state_controller to publish on the /joint_states topic (robot_state_publisher is listening to this one).

From there, what to do:

1.

Try to launch desktop_rviz_simulation.launch, but before that edit the controllers.launch file : change 
`<node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" output="screen" />`
to
`<node name="robot_state_publisher" pkg="robot_state_publisher" type="state_publisher" />`

I just found out that the executable (type="") used for the node is not the same. No idea why, I haven't checked more about it yet, but... You might get lucky.

2.

If this doesn't work, second step is to check that ros_control is correctly creating the joint_state_controller, and that this controller is running (--> publishing on /joint_states).

Run `rostopic echo /joint_states`. If you're unable to see some data (it should be published at 10Hz), it means there is a problem with the controller.

Run `rosrun controller_manager controller_manager list`, and check that you see `'joint_state_controller' - 'hardware_interface::JointStateInterface' ( running )`

---

With those tests we'll be able to locate (I hope) the problem and maybe get some answers on what to do next.

renarded, 2018-12-13:
So, it seems the error is coming from the joint_state_controller of ros_control. It's running but not working correctly.

I've checked again the logs that you provided but found no additional useful info. 

To get more info (logs), you can try to use gdb (see [here](http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB)). Maybe this will give more internal info about the joint_state_controller.

Also maybe open an issue on the ros_control github repo ? They could have already dealt with this problem.

jbiams77, 2020-02-21:
I had the same issue, it was because i didn't have joint publisher gui.
This was the fix:
[CODE] sudo apt-get install ros-kinetic-joint-state-publisher-gui [/CODE]

closed 2020-07-06 by remi2257