NiryoRobotics/niryo_one_ros #23: auto calibration infinite loop.

wyojustin, 2018-12-31:
I received my new, assembled Niryo One last week (Dec 28 2018) and everything worked well out of the box.  This morning I booted the Niryo One and requested an auto calibration from the python api.  Here is what happens next:

Joint 0 (at base of Niryo One closest to rpi) rotates slowly to near +180 degrees, then jerks back and forth a few degrees indefinitly like a limit switch is not triggered.

All other joints behave as expected.

Also, I've repeated the experiment using Niryo One Studio with the same result except I could not kill the process and had to do a remote shutdown from SSH to get it to stop.

FEATURE REQUEST: Add cancel to Auto Calibration interface.

renarded, 2019-01-02:
Hi,
The issue you have is mechanical. This is probably due to the belt not being tensed enough on the first axis.

Check out [this tutorial](https://niryo.com/docs/niryo-one/debug-your-robot/tense-the-belts/) to tense the belt. Also if it doesn't work, you might have a look at [timing pulleys](https://niryo.com/docs/niryo-one/debug-your-robot/make-sure-the-timing-pulleys-are-fixed/) and also [magnets](https://niryo.com/docs/niryo-one/debug-your-robot/fix-the-magnet-position-on-niryo-stepper-motors/) on stepper motors.

For now it's not possible to cancel an auto calibration (my bad, poor design choice from the beginning. We might look into that in the future), but there is a timeout set at 40s so the robot will eventually stop.

Let me know if this solves your issue. For further debugging you can also [contact our technical support](https://niryo.com/contact/) directly on our website.

wyojustin, 2019-01-02:
Thank you for the prompt reply.  That was the problem.   I will contact your technical support directly next time.

closed 2019-01-02 by wyojustin