NiryoRobotics/niryo_one_ros #29: ROS2 support

ZdenekM, 2019-06-03:
Any estimation when port to ROS2 might be expected? Alpha version of MoveIt seems to be released into Dashing.

renarded, 2019-06-04:
For now there is no estimation for when we'll port Niryo One code to ROS2.

Before that I guess we'll port the code from ROS Kinetic to ROS Melodic (Ubuntu 16.04 -> 18.04), which shouldn't take long and is scheduled for the next few weeks/months.

For ROS2 I think we'll wait for Moveit to be at least in a beta phase.

closed 2019-06-04 by renarded

fferri, 2020-08-30:
Hello,

MoveIt 2 is in [beta](https://moveit.ros.org/moveit/ros2/2020/02/18/moveit-2-beta-feature-list.html) and supports Ubuntu 20.04 / ROS 2 Foxy Fitzroy.

fferri, 2020-09-10:
Hello,

MoveIt 2 [has a stable (non-beta) release](https://discourse.ros.org/t/moveit-2-is-out-of-beta/16261) as of 4 September 2020.