NiryoRobotics/niryo_one_ros #45: URDF file

MADONOKOUKI, 2020-12-17:
Hi, My name is koki.

I purchase a niryo one robot. And I want to use urdf file.
In this repository, https://github.com/NiryoRobotics/niryo_one_ros/tree/master/niryo_one_description/urdf shows the urdf file. However, when calculating the inverse kinematics using this file, I received wrong result as I expected.

Can I make urdf file from ros in niryo one linux system?
I think each joint's coordinate is difficult to get. For this reason, I want to know how to get the correct urdf file from my own niryo one robot.

Best.

MADONOKOUKI, 2020-12-17:
Here is my .urdf file
```
<?xml version="1.0" ?>
<!-- =================================================================================== -->
<!-- |    This document was autogenerated by xacro from niryo_one.urdf.xacro           | -->
<!-- |    EDITING THIS FILE BY HAND IS NOT RECOMMENDED                                 | -->
<!-- =================================================================================== -->
<!--
    niryo_one.urdf.xacro
    Copyright (C) 2017 Niryo
    All rights reserved.

    This program is free software: you can redistribute it and/or modify
    it under the terms of the GNU General Public License as published by
    the Free Software Foundation, either version 3 of the License, or
    (at your option) any later version.

    This program is distributed in the hope that it will be useful,
    but WITHOUT ANY WARRANTY; without even the implied warranty of
    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
    GNU General Public License for more details.

    You should have received a copy of the GNU General Public License
    along with this program.  If not, see <http://www.gnu.org/licenses/>.
-->
<robot name="niryo_one" xmlns:xacro="http://www.ros.org/wiki/xacro">
  <!-- Properties -->
  <!-- Links -->
  <link name="world"/>
  <link name="base_link">
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="package://niryo_one_description/meshes/v2/collada/base_link.dae"/>
      </geometry>
    </visual>
    <collision>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="package://niryo_one_description/meshes/v2/stl/base_link.stl"/>
      </geometry>
    </collision>
  </link>
  <link name="shoulder_link">
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="package://niryo_one_description/meshes/v2/collada/shoulder_link.dae"/>
      </geometry>
    </visual>
    <collision>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="package://niryo_one_description/meshes/v2/stl/shoulder_link.stl"/>
      </geometry>
    </collision>
  </link>
  <link name="arm_link">
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="package://niryo_one_description/meshes/v2/collada/arm_link.dae"/>
      </geometry>
    </visual>
    <collision>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="package://niryo_one_description/meshes/v2/stl/arm_link.stl"/>
      </geometry>
    </collision>
[... 91 lines cut ...]
    <axis xyz="0 0 1"/>
  </joint>
  <joint name="joint_5" type="revolute">
    <parent link="forearm_link"/>
    <child link="wrist_link"/>
    <origin rpy="0 -1.5707963268 0" xyz="0 0 0.18"/>
    <limit effort="1" lower="-1.74533" upper="1.91986" velocity="1.0"/>
    <axis xyz="0 0 1"/>
  </joint>
  <joint name="joint_6" type="revolute">
    <parent link="wrist_link"/>
    <child link="hand_link"/>
    <origin rpy="0 1.5707963268 0" xyz="0.0164 -0.0055 0"/>
    <limit effort="1" lower="-2.57436" upper="2.57436" velocity="1.0"/>
    <axis xyz="0 0 1"/>
  </joint>
  <joint name="hand_tool_joint" type="fixed">
    <parent link="hand_link"/>
    <child link="tool_link"/>
    <origin rpy="-1.5707963268 -1.5707963268 0" xyz="0 0 0.0073"/>
    <!-- <origin rpy="-0.002 1.329 0.007" xyz="0.075 0.001 0.162"/> -->
  </joint>
</robot>

```

zakimohzani, 2021-03-09:
Hi, I'm using IKFast and some of the IKs when the end-effector are pointing down are failing. I haven't made a point cloud plot of these failures. Perhaps later.

However, I wanted to ask how is progress on your side?