NiryoRobotics/niryo_one_ros #51: standard gripper

MADONOKOUKI, 2021-04-12:
Does anyone know how to add a standard gripper on niryo one robot at the rviz visualization?

ValentinPitre, 2021-09-17:
Hello, 
You can follow this tutorial to display some markers with a "mesh_resource". The STL files are available here: https://github.com/NiryoRobotics/niryo_one/tree/master/STL/7-Tools/Gripper1

hielsnoppe, 2022-07-15:
We created simulation models for the standard gripper based on the above 3D models. The other tools are work in progress. You can get them from my fork at https://github.com/hielsnoppe/niryo_one_ros/tree/tools_support

To launch in RVIZ run `roslaunch niryo_one_tools_support display_gripper_1.launch`.

Note, however, that this is work in progress. Feel free to contribute and improve!