NiryoRobotics/niryo_one_ros #53: Python 3 support

mrkrizic, 2021-05-18:
Hello, is there a way to use this API with python 3 as the support for python 2.7 was dropped earlier this year. If not are there any plans to upgrade this repository to be compatible with python 3?

ValentinPitre, 2021-09-17:
Hello ! 

You can control the Niryo robot through the PyNiryo API which is compatible with both Python 2.7 and Python 3.x. : https://docs.niryo.com/dev/pyniryo/v1.0.4/en/index.html

`pip install pyniryo`

Full compatibility with Python 3 requires a transition from ROS Kinetic to ROS Noetic.

mrkrizic, 2021-11-17:
Hello there,

I am a bit confused. Currently I am still using the NiryoOne library running on the robot. Since my computer is already running multiple programs in parallel, I would like to be able to have my scripts on the robot itself.

The library you provided shows me how to control the robot remotely. Since I am not doing that, I need to use this ROS wrapper which is only compatible with python 2 (for which support has dropped in january 2020).

I think the only thing that prevents this library from being compatible with python 3 are  Exception block are two lines:
221 and 229 in the niryo_one_api.py script both use
`except rospy.ROSException, e:
                raise NiryoOneException(e)`

If those two lines would be changed to match the syntax of the third exception in line 291:

` except rospy.ROSException as e:
                    raise NiryoOneException(str(e))`
                    
Then the entire api would be python 3 compatible. However, I don't know whether this could then be actually run on a Niryo One robot.

Is there a way to resolve this issue?

ValentinPitre, 2021-11-18:
In reality the version of python used in the robot is dependent on the ROS version and Ubuntu.Currently the robot is running Ubuntu 18, so ROS Melodic. ROS Melodic works with Python2 only.  In reality when you use the ROS API of the robot, you use rospy, so python 2. 
The way to get around this is probably to program via the PyNiryo API with the local ip 127.0.0.1. And send this code to the robot.

Translated with www.DeepL.com/Translator (free version)

mrkrizic, 2022-01-03:
I have tried in the simulation and it does not work there. I won't be investigating on the robot myself any further as I may break something by accident. The new API is solely designed for the NED and is useless on the NiryoOne.

Why have there been no efforts to upgrade the ROS and python version? (Except maybe this community pull request from half a year ago: https://github.com/NiryoRobotics/niryo_one_ros/pull/54) It has been two years since the support for python2 has been dropped! This should have already happened more than 2 years ago.

ValentinPitre, 2022-01-07:
In fact the NiryoOne product is no longer on the market, so no transition to a newer system will be made.

The production of the NED will also be stopped in favour of the NED2 which will be available on newer systems.

A final update will be done in mid January to restore pyniryo compatibility with the Niryo One

mrkrizic, 2022-01-10:
Ok so I will be able to use the pynyrio package for the NiryoOne, but I would still have to use a python2 interpreter. Did I understand this correctly?

ValentinPitre, 2022-01-24:
Pyniryo  is compatible with both Python 3 and 2 :
https://docs.niryo.com/dev/pyniryo/v1.1.0/en/source/setup/installation.html

mrkrizic, 2022-01-24:
Thank you, I will definetly try it out

closed 2022-01-24 by mrkrizic