NiryoRobotics/niryo_one_ros #61: move_pose() is not working.

zhaoza, 2022-09-22:
I am trying the pick and place sample program during the simulation.  I can't use the move_pose() or pose functions. The error is as follows: (move_joints works fine)

_Traceback (most recent call last):
  File "test1.py", line 49, in <module>
    pick_n_place_version_2(client)
  File "test1.py", line 24, in pick_n_place_version_2
    robot.move_pose(pick_pose_high)
  File "/home/robotics/.local/lib/python3.6/site-packages/pyniryo/api/tcp_client.py", line 472, in move_pose
    self.__send_n_receive(Command.MOVE_POSE, *pose_list)
  File "/home/robotics/.local/lib/python3.6/site-packages/pyniryo/api/tcp_client.py", line 152, in __send_n_receive
    return self.__receive_answer(with_payload=with_payload)
  File "/home/robotics/.local/lib/python3.6/site-packages/pyniryo/api/tcp_client.py", line 138, in __receive_answer
    raise NiryoRobotException("Command KO : {}".format(received_dict["message"]))
pyniryo.api.exceptions.NiryoRobotException: Command KO : 6 parameters expected but 7 given
In function_

ValentinPitre, 2022-12-06:
Hello, you don't have compatible versions of robot/pyniryo. For older versions, please use version 1.0.5 of pyniryo. The other way to get all the latest features is to [update your robot.](https://niryo.com/download)

See the compatibility table: 
https://docs.niryo.com/dev/pyniryo/v1.1.2/en/index.html?lang=fr

Gvny, 2023-01-10:
Thank you for your response. 
I've tried both. 
First: intalling pyniryo 1.0.5 and python 3.5 instead of the 3.11 I was using. 
But even the line "from pyniryo import* " does not work. The version seems to be too old..
So, I came back to the 3.11 version and
 then trying to update the robot which characteristics are bellow: 
[image]

I connected to the ned through a wifi network. Once the connection is done, the ned led turned green and a pop - up displayed and showed updates are available. But when I click on the update button, it redirect me to this page (https://docs.niryo.com/product/niryo-studio/v4.1.1/en/source/settings.html#ned-software-update) and no update starts..

Can you suggest something to achieve the update? Or to have a good versions alignment according to the current robot status on the snapshot attached before? 
Thanks

ValentinPitre, 2023-01-11:
You can also update your robot manually:

-  Unscrew the screws located on the top of the base for NED or just simply watch under the robot for the Niryo One
- Remove the SD card of the raspberry pi
- Download the latest image version on this website: https://niryo.com/download
- Flash the SD card with this image with a flashing software like Balena etcher 
- Replace the SD card and restart

For the Niryo One, you have a tutorial here: https://niryo.com/docs/niryo-one/update-your-robot/update-raspberry-pi-image/

The-Kriz, 2023-03-13:
I had the same issue:

The reason for this issue is due to installation of different PyNiryo version 

in my case:

[image]

Robot Software version is 4.0.1

And the PyNiryo version installed at the time was 1.1.2
[image]

These 2 different versions does not work together 

https://docs.niryo.com/dev/pyniryo/v1.1.2/en/index.html

PyNiryo version | ROS Stack version | Robot
-- | -- | --
<= 1.0.5 | <= 3.2.0 | Ned
1.1.0 | 4.0.0 | Niryo One, Ned, Ned2
1.1.1 | 4.0.1 | Niryo One, Ned, Ned2
1.1.2 | >=4.1.1 | Niryo One, Ned, Ned2

</body>
</html>

So i reinstalled my PyNiryo version to 1.1.1 from 1.1.2 (which is an easy fix)
or you can just upgrade the ROS Stack version 

Hope this will fix your issue